hephy-dd/diode-measurement #14: unclear delay between "start" button pressed and IV ramp starts

thomasbergauer, 2021-12-19:
the title says it all. The feeling is also that it lasts much longer on the first start after the software was opened. But even later it takes several seconds until something happens after the start button was pressed.

If it is intentional then it needs to be written somewhere how long this delay lasts.
If it is not intentional then we need to investigate why it takes so long.

arnobaer, 2022-01-11:
It turned out that creating a pyVISA resource using GPIB bus can take up to several seconds.

This effect can be reproduced using a simple script like following:
```python
import logging
import pyvisa

logging.basicConfig(level=logging.DEBUG)  # enable pyVISA debug messages

rm = pyvisa.ResourceManager()  # using NI VISA driver

with rm.open_resource('GPIB0::11::INSTR') as instr:  # might take several seconds
    instr.query('*IDN?')
```

thomasbergauer, 2022-01-11:
So is this an issue with the NI-VISA/GPIB driver? Any parameter to be optimized there?

thomasbergauer, 2023-03-13:
This is an issue with the GPIB-Ether interface, as no delay is observed when the device is connected via USB (and Visa configured accordingly). Thus I close here.

closed 2023-03-13 by thomasbergauer