hephy-dd/diode-measurement #34: Timeout error on start

hao-yin-2609, 2022-05-09:
Setup:

1. sim SMU: `python -m comet.emulator.keithley.k2410 -p 11002`
1. on start the communication works:
```
INFO:root:send (54, 'b'Spanish Inquisition Inc., Model 2410, 43768438, v1.0\r\n'')
INFO:root:recv (13, 'b':OUTP:STAT?\r\n'')
INFO:root:send (3, 'b'0\r\n'')
INFO:root:recv (26, 'b':SOUR:VOLT:LEV 0.000E+00\r\n'')
INFO:root:recv (7, 'b'*OPC?\r\n'')
INFO:root:send (3, 'b'1\r\n'')
INFO:root:recv (6, 'b'*CLS\r\n'')
INFO:root:recv (7, 'b'*OPC?\r\n'')
INFO:root:send (3, 'b'1\r\n'')
INFO:root:recv (17, 'b':ROUT:TERM FRON\r\n'')
INFO:root:recv (7, 'b'*OPC?\r\n'')
INFO:root:send (3, 'b'1\r\n'')
INFO:root:recv (24, 'b':SOUR:FUNC VOLT\r\n*OPC?\r\n'')
INFO:root:send (3, 'b'1\r\n'')
INFO:root:recv (33, 'b':SENS:CURR:AVER:TCON MOV\r\n*OPC?\r\n'')
...
```
1. The exception occurs at some time after the start command (few secs):
```
2022-05-09T09:50:20::diode_measurement.measurement::ERROR::K2470: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Traceback (most recent call last):
  File "diode_measurement\resource.py", line 56, in read
  File "pyvisa\resources\messagebased.py", line 486, in read
  File "pyvisa\resources\messagebased.py", line 442, in _read_raw
  File "pyvisa_py\highlevel.py", line 519, in read
  File "pyvisa\highlevel.py", line 251, in handle_return_value
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "diode_measurement\driver\driver.py", line 12, in handle_exception
  File "diode_measurement\driver\k2470.py", line 76, in _query
  File "diode_measurement\resource.py", line 43, in query
  File "diode_measurement\resource.py", line 60, in read
diode_measurement.resource.ResourceError: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "diode_measurement\measurement\__init__.py", line 110, in run
  File "diode_measurement\measurement\__init__.py", line 328, in measure
  File "diode_measurement\measurement\iv.py", line 54, in acquireReading
  File "diode_measurement\measurement\iv.py", line 34, in acquireReadingData
  File "diode_measurement\driver\k2470.py", line 64, in read_current
  File "diode_measurement\driver\driver.py", line 14, in handle_exception
diode_measurement.driver.driver.DriverError: K2470: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2022-05-09T09:50:24::diode_measurement.measurement::ERROR::K2470: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Traceback (most recent call last):
  File "diode_measurement\resource.py", line 56, in read
  File "pyvisa\resources\messagebased.py", line 486, in read
  File "pyvisa\resources\messagebased.py", line 442, in _read_raw
  File "pyvisa_py\highlevel.py", line 519, in read
  File "pyvisa\highlevel.py", line 251, in handle_return_value
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "diode_measurement\driver\driver.py", line 12, in handle_exception
  File "diode_measurement\driver\k2470.py", line 76, in _query
  File "diode_measurement\resource.py", line 43, in query
  File "diode_measurement\resource.py", line 60, in read
diode_measurement.resource.ResourceError: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "diode_measurement\measurement\__init__.py", line 118, in run
  File "diode_measurement\measurement\__init__.py", line 365, in finalize
  File "diode_measurement\measurement\__init__.py", line 354, in read_source_voltage
  File "diode_measurement\driver\k2470.py", line 67, in read_voltage
  File "diode_measurement\driver\driver.py", line 14, in handle_exception
diode_measurement.driver.driver.DriverError: K2470: TCPIP0::localhost::11002::SOCKET: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

hao-yin-2609, 2022-05-09:
note this error only occurs at v0.9.2
same behavior with rpc... works with v0.8.0

arnobaer, 2022-05-17:
The problem occurred due to a wrong driver selection (running a K2410 emulator and selecting K2470 as driver).

closed 2022-05-17 by arnobaer