hephy-dd/diode-measurement #104: CLS resets Initialization of Brandbox Switch

prakhub, 2024-08-02:
After the initializing the switch

https://github.com/hephy-dd/diode-measurement/blob/ef06b805224ebbf3d82fb0fa0bd57a5ab1c4a9ed/diode_measurement/measurement/__init__.py#L344
The clear state signal is sent to the Switch Box.
https://github.com/hephy-dd/diode-measurement/blob/ef06b805224ebbf3d82fb0fa0bd57a5ab1c4a9ed/diode_measurement/measurement/__init__.py#L356
However, maybe unexpectedly, the clear state command also changes the switch states and sets the switchbox to the default I-V mode.
```
*CLS - Clear status (Relays are set to IV mode)
```
In the default "I-V" mode, some switches are already closed.

Later, in  the configure action, only the switches to be closed are sent as a command:
https://github.com/hephy-dd/diode-measurement/blob/ef06b805224ebbf3d82fb0fa0bd57a5ab1c4a9ed/diode_measurement/driver/brandbox.py#L56 

This means that the switches which are closed via `*CLS` are never opened again and the configuration of the switch box is wrong.

prakhub, 2024-08-02:
In my opinion, there are two possible fixes:

- Move the `initialize_switch()` command after "Clear State"
- Or (better), also expclitely open the channels which should be open during the "Configure" step https://github.com/hephy-dd/diode-measurement/blob/ef06b805224ebbf3d82fb0fa0bd57a5ab1c4a9ed/diode_measurement/driver/brandbox.py#L56

arnobaer, 2024-08-05:
So this is unexpected! Maybe the better option to overcome this behavior would be to explicitly call `open_all_channels()` in the BrandBox drivers `configure()` method:
```python
def configure(self, options: dict) -> None:
    self.open_all_channels()
    channels = options.get("channels", [])
    self.close_channels(channels)
```

PR #105 (2024-08-07): 0.21.1
fixes #104
closed 2024-08-07 by arnobaer