hephy-dd/diode-measurement #107: Datapoints in Graph disappear at very high zoom levels

prakhub, 2024-11-06:
When manually zooming in very close to the data points, they seem to disappear and stop being displayed at a certain zoom level.
[image]
The first image shows some I-V data at a "normal" zoom level.
[image]
In the second image, the graph is zoomed in a bit further, however, no data points remain visible (although they should be inside of the axis limits).

At first I thought this was an issue with the axis range being limited to 1 pA, see https://github.com/hephy-dd/diode-measurement/blob/97dfb6b89da59d859213f5d839ec8b87788e5f6e/diode_measurement/view/plots.py#L258 but if I understand correctly, this only matters for the auto-scaling of the graph, not for manual zooming.

Maybe this is an Issue with the PyQt chart? Or some floating point errors?

arnobaer, 2024-11-06:
This is a known issue with QtChart. It can't handle very small float numbers.

prakhub, 2024-11-07:
I see.
I guess we could implement some kind of workaround by scaling up the numbers in the QtChart, but probably it is not worthwhile, as zooming in that far is not often done.

arnobaer, 2024-11-08:
Indeed I used scaling of values in other software but it is quite tricky to adjust the dynamic axis then. Switching to [PyQtGraph](https://www.pyqtgraph.org/) would be an alternative but comes with a bunch of other downsides.